uofw/uofw #43: Update to 6.61

artart78, 2015-01-17:
A new PSP firmware version has been released!

I've checked the differences between 6.60 in detail. Here they are:
- added 05g (PSP Go) stuff [encrypted, cannot compare to 6.60]
- no more reboot_{07,09,11}g.bin?
- new certificates
- new version number in many modules (in module info)
- changes in cert_loader.prx [inside code]
- changes in pspbtcnf*.bin [version number]
- changes in lflash_fatfmt.prx [inside code]
- changes in libhttp.prx [inside code]
- changes in libssl.prx [MANY changes inside code]
- changes in loadexec_*.prx [probably only reboot.bin + addresses because size changed]
- changes in *_files_table.bin (psardump files)
- changes in nandipl_*g.ipl [probably mostly related to other modules changing]
- changes in reboot*.bin [probably mostly related to other modules changing]
- changes in vsh/etc/index_*.dat
- changes in vsh/etc/version.txt
- changed version number in init.prx [in devkit version printing]
- changed version number in loadcore.prx [CheckDevkitVersion()]
- changed version number in power_*g.prx [number in sub_00001AE8(), in structure passed to scePower_driver_C463E7F2()]
- changed version number in sysmem.prx [sceKernelDevkitVersion() return value], lfatfs.prx [sceLFatFs_driver_F28896C0() return value]
- changed version number in htmlviewer_plugin.prx [twice]
- changed version number in htmlviewer_utility.prx [looking like one of the numbers in htmlviewer_plugin.prx]

That'd need no change in uOFW (just updating SDK_VERSION to 0x06060101, since the really modified modules haven't been RE'ed yet), but I wanted to share it with you. :) It's quite disappointing that the project is quite dead now, but I guess everybody's been too busy, and RE'ing is a long and hard task.

joel16, 2015-01-17:
Funny, the 6.61 version.txt says the firmware was made all the way back in November 13, 2014. Well that was the date supplied so I'm guessing they have been working with this a while back.

artart78, 2015-01-17:
I think PSP isn't really one of their priorities. I guess someone made the changes, and they only decided to really push them now.

caffinatedmonkey, 2015-01-18:
I think one of the reasons this project is dead, is because there is no motivation.

joel16, 2015-01-18:
Oh c'mon I've been motivating you guys from the very start :( I've even wrote an article on wololo.net/talk dedicated to you guys. And I'm always checking up here to see new commits, sadly there hasn't been any recently. I'm willing to write another one again to get you guys more support.

Omega2058, 2015-01-19:
It's perfectly fine joel and I feel the same way. Initially I had lots of motivation, but as time went on it just slowly dwindled down.

qwikrazor87, 2015-01-19:
Do you guys have plans of reversing the OFW updater modules (I know it's not part of the modules in flash)?

Omega2058, 2015-01-20:
As far as I know, I don't think that's planned. But I personally don't see why it shouldn't be added to the list of things to do in the future.

artart78, 2015-01-20:
Well, personally, I lost my free time more than my motivation. But clearly, RE'ing is a very long process, and I think no-one had enough free time to continue working on uOFW.

Reversing the updater modules is not one of our plans, as our goal is to fully understand what the firmware does and how it works along with the hardware. As the updater probably just copies files or contains embedded parts of other modules, I don't think it's worth it.

libcg, 2015-01-20:
Thanks for the changelog, @artart78. This is really interesting.

I'll be working on idstorage.prx this week. I really want to finish it, it's just a matter of fixing bugs to get it running now.

qwikrazor87, 2015-01-20:
@artart78
Yeah, I just thought it would be interesting though, since there are some interesting modules in there (to me at least).

modname (attr)

updater	(0x0800)
scePSAR_Driver (0x1006)
sceChkuppkg (0x5006)
sceTexureLoader (0x1007)
sceSuspendCaneler (0x1007)
SetDisplayBuffer (0x1000)
TexSeqCounter (0x1007)
sceUpdate_driver (0x1007)
SyncProgressBar (0x1007)
sceNetworkUpdate (0x0800)
sceCheckDegeneration (0x1007)
sceNAND_Updater_Driver (0x1007)
sceLFatFs_Updater_Driver (0x1007)
sceLflashFatfmtUpdater (0x1006)

For the last three there are two of each, with tag 0x00000000 and 0x4CXXXXXX (can't remember).

joel16, 2015-01-22:
Hey guys just to let you know, I'm willing to write another article about this because its been a while since I've done that. And no it's not just to motivate you guys but to also hopefully catch other developers attention. I'm gonna be listing all your names as the current list of developers. Please do notfiy me if I'm missing anyone.
- artart78 
- Omega2058
- libcg
- qwikrazor87
- Felix-Dev
- Kern--
Once again sorry for going offtopic, if you wish to move this discussion elsewhere please do let me know.

devnoname120, 2015-03-19:
@joel16 I don't know if you wrote the article, but I think it might not be this useful. There are very few PSP developers left, so we don't (and we cannot) get enough reverse engineers for getting interesting results. The last module which was worked on (module_mgr by @Felix-Dev and @devnoname120) has not been updated for a while (although it's near completion), and we both have lost motivation for finishing it, for all I know.

joel16, 2015-03-20:
Aw c'mon guys there is no rush. Take all the time you need. 5 more years? sure. Like artart said RE'ing is a very long process. You guys do deserve the motivation for coming this far. Even some vita developers have been reversing psp stuff, so I'm sure a few would be interested. I haven't written the article as yet cause I've been very busy recently. I will try to finish it by tonight. I would be more than happy to contribute to the project, I just need someone to point me in the right direction. I need to learn about asm though, and I'm planning to do that after my exams (June).
PPSSPP and other Vita developers have also been using this as a source for information, so what you guys have done really is helping.

hlide, 2015-05-11:
Hey people,

I'm looking for more contributors for [pspe4all](https://github.com/psce4all/psce4all), especially in the firmware field. I used to reverse engineered firmware as I needed to cross-analyze it with Utopia and uOFW results. Originally, the pspe4all close source was closer to what firmware modules do (threadman, sysmem, ge, etc.). Right now, I want to focus my efforts into x86-64 interpreter-like dynarec and so I need more people for coding HLE part, so if you are interested, just knock.

Interpreter-like mode means generating X86-84 instructions per Allegrex instructions (or two when a delay slot occurs). Some details [here](https://github.com/psce4all/psce4all/wiki/X86-64-Dynamic-Recompiler-using-a-huge-instruction-cache).

Here is how pspe4all in interpreter-like mode performs: [PSP-CompilerPerf-Benchmarks]( https://github.com/psce4all/psce4all/wiki/PSP-CompilerPerf-Benchmarks)