uofw/uofw #45: LAN.st alternatives

simar7, 2015-06-20:
Hello!

I was going through the README.md and seems like LAN.st has been down for a while. Any alternatives for it as the guides it had were quite helpful in understand PSP knowledge.

caffinatedmonkey, 2015-06-20:
There are some caches in the way back machine but they aren't great. http://web.archive.org/web/20131002210745/http://lan.st/forumdisplay.php?s=a5d34385b0fb8918fc8696cfe64ad761&f=83
I wish the original creators of this content were more careful.

simar7, 2015-06-20:
@caffinatedmonkey thanks, I did look that before asking here but it seems that the PSP tutorials aren't archived. :disappointed:

Felix-Dev, 2015-06-21:
I uploaded the linked pages on my fork of the uOFW project, you can find them in the ReadMe file. We can think about a nice way to integrate them into the uOFW project.

https://github.com/Felix-Dev/uofw/blob/master/README.md

simar7, 2015-06-21:
@Felix-Dev thanks!

I'm also thinking that [readthedocs](https://readthedocs.org/) could be a possibility to organize information better. This could be solely used for PSP research data, even for those not directly involved in uOFW effort. Could benefit community as a whole.

Felix-Dev, 2015-06-23:
Here is a first version showing how the uploaded data can be organized: http://felix-dev.github.io/uofw/docs/

Since most of the data uploaded are either PDFs or HTML-pages, I currently think using Github as the host platform is sufficient. Let me know what you think.

ludkiller, 2015-06-23:
what kind of info are you looking for @simar7  ?

simar7, 2015-06-23:
@Felix-Dev looks good to me!

caffinatedmonkey, 2015-06-28:
This is amazing! @Felix-Dev Can you please put all your resources from the psp hacking scene into a repository so it doesn't get lost in the past? :100:

Felix-Dev, 2015-06-29:
PSP research data not collected by uOFW can now be found here: http://uofw.github.io/upspd/docs/

I believe our upspd-repository is a good place to hold the collected data as we intend to make it the central place for PSP documentation. The given data can be easily updated by pull-requests even by people not working on the uOFW project.

Alongside this update, I also updated uOFW's readme (as well as certain parts of the wiki). I think the new readme now guides newcomers more directly to central uOFW information.

Thanks for your feedback and keep it coming, guys!

@caffinatedmonkey: The documents uploaded right now cover the more useful parts of my PSP hacking resources. Of course I will upload more information when I deem it useful.

simar7, 2015-06-29:
@Felix-Dev this is awesome! thanks!

closed 2016-06-19 by devnoname120