uofw/uofw #50: Not an issue but a presentation of snowman for allegrex (WIP)

hlide, 2015-08-07:
Hi,

I'm working to integrate *Allegrex* support in *snowman* decompiler. While working on it, I will be glad if some other people here may contribute in *snowman*, especially in the field of binary parser like PRX module. Or even trying to test it with ELF file to check what kind of issues or improvements *snowman* for *Allegrex* may need.

The place is [here](https://github.com/hlide/snowman) and the topic where it starts is [here](https://github.com/yegord/snowman/issues/36).

Regards.

artart78, 2015-08-07:
Hey hlide,
I won't have time to work on PRX file loading the following weeks, and I'm not sure I'll have the time later either. However, it shouldn't be very hard using PRXTool: everything is in ProcessPrx.C, and I think the only functions you really need since standard ELFs are already supported are LoadRelocs() (especially LoadRelocsTypeB()), maybe FixupRelocs(), and mostly CreateFakeSections() which creates fake sections corresponding to the PRX segments.

hlide, 2015-08-07:
Hey Arthur,

indeed, it seems I need to integrate a big part of prxtool (including xml file for symbols and syscalls) to get better signature for all external functions.

nihilus, 2015-08-07:
Is it supported by libbfd?

hlide, 2015-08-07:
prxtool and libbfd are separate things. Prxtool is a standalone program to disassemble, extract PRX file into ELF, generate IDA scripts, generate a XML file containing symbols with their NIDs, and so on. As for libbfd, there is probably a patch for Allegrex somewhere but I think it is a very old version.

nihilus, 2015-08-07:
It would be nice to integrate libbfd... I think it would make more sense than building own parsers. Building our own parsers was something I regret with boomerang.SF.net

nihilus, 2016-01-14:
Now Ive implementera parsing prx files via libbfd for snowman.

zecoxao, 2016-04-21:
hi.
@nihilus i have decrypted the psp firmware and compiled your latest repository revision, but snowman cannot recognize the instructions, and decompilation results in no output.
am i missing something?

nihilus, 2016-04-21:
did you use IDA, libbfd- or the ELF-parser. Libbfd has a bug for prx-files without sections.

zecoxao, 2016-04-21:
i didn't use IDA or libbfd. and if libbfd has bugs i'll probably not use it. do you mean by ELF parser the default parser used by Snowman?

nihilus, 2016-04-24:
@hlide is fixing up the relocation format and program header in the ELF-loader sufficient? I got some files from @zecoxao which seems to need the prxloader plugin for IDA Pro.

zecoxao, 2016-04-24:
https://www.sendspace.com/file/z8vurd leaving the source here in case @hlide wants to take a look

zecoxao, 2016-04-25:
https://gist.github.com/anonymous/b097cc844cde21e501916209e72082ad
this is the file we modified in order for *some* prx modules to work with the parser. 
take into consideration that using allegrex crashes the system, while using mips only crashes modules with delay slots or invalid instructions (and the program if you try to click on it on a prx without these, don't ask me why though)

nihilus, 2016-04-25:
@zecoxao since it wont recognize the program header and hence the section table it will try to disassemble things that shouldn't be disassemble such as text segments and will generate bogus assembly code with weird delay slots.

hlide, 2016-04-25:
Sorry, I have little time because of my job. My best hint is to use the prxtool stuff into Snowman for Allegrex.

hlide, 2016-04-25:
Snowman suffers some misconceptions in the way to handle some expressions. For instance, it tends to think memory access can be reordered and be discarded while I'm strongly against this behavior.

hlide, 2016-04-25:
Mine is using its own but faster Allegrex disassembler and nihilus' probably using Capstone Engine which is not good for Allegrex disassembly in my opinion. But Snowman lacks of intrinsic function expression to handle special instructions (especially VFPU and COP instructions)

zecoxao, 2016-04-29:
i've been talking with @nihilus and aerosoul94 and it seems nihilus latest decompiler handles most elfs generated by prxtool -e option, thanks to their help.
the latest binary (as of now) is here: https://www.sendspace.com/file/ncz0ra
non working elfs are specified in the txt

zecoxao, 2016-05-29:
@joel16 asked for this so...
https://www.sendspace.com/file/3jteqm
snowman IDA plugin (should support IDA 6.5-6.8)

joel16, 2016-05-29:
Sorry about not getting back to you on wololo. I got held up with other things. I recently bought a new laptop, so getting all of this setup is gonna take a while. Thanks for this :)

closed 2021-01-19 by artart78