uofw/uofw #51: SysMemForKernel_7FF2F35A

joel16, 2015-08-12:
https://github.com/uofw/uofw/blob/e018ffeac1b8048703cdc20aee3f590ffad465e9/src/sysmem/exports.exp#L59

Line 59
PSP_EXPORT_FUNC_NID(SysMemUserForUser_945E45DA, 0x7FF2F35A)
Is this correct? Function NIDs are always the same as their function names (From what I've seen atleast)
The NIDs 0x7FF2F35A refers to SysMemForKernel_7FF2F35A from what I know.

qwikrazor87, 2015-08-12:
ever since the late 3.XX firmwares Sony has been "randomizing" the majority of the kernel NIDs, that's why CFW has a NID list for supported firmwares, that way we can search for a fixed NID (NIDs are usually used from 3.71 before Sony started changing kernel NIDs) and not worry about the current firmware NID.

joel16, 2015-08-13:
Yeah Felix91 and artart78 told me on irc. I never knew this, maybe I read about it ages ago but had forgotten it. Thanks anyway :)

closed 2015-08-13 by joel16