uofw/uofw #53: Hybrid approach (mix of HLE and LLE) for a PSP emulator

hlide, 2016-06-26:
I may resume my work on my psp emulator, but with a "small" change in my plans: going into an hybrid approach between HLE and LLE for modules including kernel ones. My plan is to let the recompiler to handle most of kernel codes instead of using a native code by keeping in a HLE state only the NID functions dealing with direct accesses to MMIO regsiters for instance.

So I post here because I may need some help to understand how the firmware is booting and installing its modules so I can have a rough idea which modules (or libraries per module) I can really keep in a LLE state.

Thanks

artart78, 2016-06-26:
I think no one has enough knowledge to know exactly that. However, I think you can get a good approximation by just running prxtool on all the .prx files and checking which ones access an address 0xBxxxxxxx; I did that some time ago, I cannot find the files again though.

hlide, 2016-06-26:
Hi Arthur,

That's the easy part. The part which bugs me is how the firmware is loaded and consolidated. What does the first sequence load and how the module manager is taking into action if it is not the first sequence?

hlide, 2016-06-26:
My intention is to let the thread manager and the module manager of the firmware to handle that stuff instead of rewriting them into native code (I noticed threadman is not yet decompiled so far here)

artart78, 2016-06-26:
There is some of such information here: https://github.com/uofw/upspd/wiki/Firmware-boot
The rest should be easy to guess using module dependencies.

hlide, 2016-06-26:
In this file [hle.zip](https://github.com/uofw/uofw/files/333551/hle.zip), I have modules.txt which says the order of modules and it is close to your list you give above. I have also already get the files "dependency.txt" for each module. In fact, I have a tool which generates all that stuff including header and source files not included here.

Back to your link. So it seems there is two parts. The first part LoadCore then Init. All the modules from sysmem.prx up to init.prx excluded are handled by LoadCore. The rest of modules are handled by Init.prx according to the choosen environment: VSH or Game. Right?

So I basically need to see how reboot.bin is proceeding to load sysmem.prx then loadcore.prx (I suppose this one will offer some basic modules handling).

hlide, 2016-06-26:
I guess I can find reboot.bin in https://github.com/uofw/uofw/tree/master/src/reboot

closed 2021-01-19 by artart78