uofw/uofw #59: Controller API update

Felix-Dev, 2020-12-29:
A bunch of APIs of the Controller module should be updated to their presumed correct namings (based on PS Vita SDK documentation). These APIs include the following which for now have custom names:

| Current Name | Proposed Name|
|---|---|
| [SceCtrlDataExt](https://github.com/uofw/uofw/blob/master/include/ctrl.h#L98)| SceCtrlData2|
| [sceCtrlPeekBufferPositiveExtra](https://github.com/uofw/uofw/blob/master/include/ctrl.h#L578) | sceCtrlPeekBufferPositive2 |
| [sceCtrlPeekBufferNegativeExtra](https://github.com/uofw/uofw/blob/master/include/ctrl.h#L592)| sceCtrlPeekBufferNegative2 |
| [sceCtrlReadBufferPositiveExtra](https://github.com/uofw/uofw/blob/master/include/ctrl.h#L606)| sceCtrlReadBufferPositiv2 |
| [sceCtrlReadBufferNegativeExtra](https://github.com/uofw/uofw/blob/master/include/ctrl.h#L620)| sceCtrlReadBufferNegative2 |

These APIs are used to obtain input data provided by a wireless controller (such as the DS3).

Felix-Dev, 2020-12-29:
Awesome to see you guys still around @joel16 and @artart78! Any kind of platform (like Discord (they have a PSP server btw, discovered it today :p) or other chat platforms) you are active these days?

artart78, 2020-12-29:
It's rare to receive a notification for this repo, hard to miss it. ;p
I'm still using IRC a lot (including #uofw which as you could guess is not very active), and also using Discord lately (I guess you can find me by my nick, or just send me an email)

Felix-Dev, 2020-12-29:
Haha, I was about to mention the IRC server but then I thought, no way does that channel still exist. Turns out I might have to pay it a visit again 😁

joel16, 2020-12-29:
Hey! I'm not all that active but yes I'm still around. I'm also part of that discord too if you're talking about the right one. Would be nice to get this going again :)

Felix-Dev, 2020-12-29:
@joel16 I'm talking about [this discord server](https://www.reddit.com/r/PSP/comments/7x28b5/join_the_psp_discord_server/). I just joined today so I'm the last new member in the #new-users channel ;)

@artart78 I sent you an email!

joel16, 2020-12-29:
I don't think that's the right one @Felix-Dev. You can find a lot more active devs as well as me here: https://discord.gg/PVBqwtmD
Also I believe this one is the biggest PSP discord server :)

Felix-Dev, 2020-12-29:
@joel16 Joined that server :)

Felix-Dev, 2020-12-31:
I just checked the NIDs for the proposed names and since they are exported in a user library as well (sceCtrl) their NIDs are only determined via their SHA1 name hash and not an additional random seed. (Not sure if these seeds are known by now but it doesn't matter here).

The SHA1 hash for the string "sceCtrlReadBufferPositive2" is `0B039810B19D4F2FFF3DB1C7160FDC9C1F29346E`. If we take the first 32 bits and represent them in the Little Endian format, we get the following value: `1098030B`.

And the NID of that function is: https://github.com/uofw/uofw/blob/d5bf38888e1b86c1d1c9506020cb364b386d45be/src/ctrl/exports.exp#L14

As we can see, they match. (This is also the case for the other three proposed function names.)

PR #64 (2021-01-01): Controller module: Updated API naming and improved documentation.

## Description
This PR updates the names of select controller module APIs to the official Sony names.

## Motivation and Context
Closes #59 

## How Has This Been Tested?
Compiled the ctrl module successfully.
closed 2021-01-01 by Felix-Dev