uofw/uofw #85: Controller module: Test if DualShock3 input is obtained correctly

Felix-Dev, 2021-03-06:
Back in 2014, it was discovered that our controller module does not properly read DualShock3 input on the PSP Go. Since then, fixes have been made to our controller implementation (i.e. commit cb3abcdb91d3db7a99ee05db3923779886b053fe) specific to reading data input from external controllers.

We should check if these fixes have solved the reported bug. Unfortunately, I myself don't have a PSP Go (which is the only PSP model which can offcially be connected with a DualShock 3 contoller), so I am hoping somebody else can test this.

Test environment required:
* PSP Go 6.60 (running CFW)
* DualShock 3 controller

Tests to perform:
* After connecting the DualShock 3 controller with the PSP Go, try using the controller to navigate through the XMB or play games with it.

ste2425, 2026-10-05:
This isn't testing on a GO, but on a 2K. However both myself and Crozone have been able to inject our own controller data using these methods.

Crozone's implementation is here:  https://github.com/crozone/PSP-EmulatedControllerTest

My implementation (which leaves allot to be desired) which was based on Corzone's is here: https://github.com/ste2425/PSP-Bluetooth-Plugin/blob/develop/plugin/controllerPatching.c I inject external controller data via SIO using an ESP32 to allow for Bluetooth controllers.

I know this isn't exactly what you was asking for but I think its a step in the right direction to confirming that you are reading the controller data correctly. 

EDIT: i can successfully inject controller data and play with this method in the XMB and in GAME. In POPS however my data does not seem to be applied out of he emulation buffer's and bridged into the main buffer. POPS uses `sceCtrlPeekBufferNegative`. If i hook `sceCtrlPeekBufferNegative` and get a reference too `sceCtrlPeekBufferNegative2` and call it in the none 2 variant i can see my data is there. 

Im still doing some more investigation into the POPS sides of things, but XMB and GAME work fine.

ste2425, 2026-10-11:
so the POPS issue turned out to be `popsman`. On my 2K at somepoint it calls `sceCtrl_driver_6C86AF22(0)` which disables the bridge between the `SceCtrlInputDataTransferHandler` fed buffer and the global buffer.

I couldn't seem to hook `sceCtrl_driver_6C86AF22` to overwrite it. Maybe my plugin runs too late, but more likely just a me issue.

So in my `SceCtrlInputDataTransferHandler` handler i just call it setting it to `SCE_CTRL_PORT_DS3`. It's a hack but it enabled external input when in POPS so i would say from my POV its working.